mooshim/Mooshimeter-AndroidApp #16: Loading Mooshimeter-AndroidApp

jkalanda, 2015-04-02:
Hey folks - 
I'm running into issues when trying to import the Mooshimeter-AndroidApp package into Android Studio. I would profoundly appreciate if you could provide some guidelines. 
I tried re-arrangeing directories in many different ways, but still no luck.

Thank you in advance

jwhong, 2015-04-05:
Hi jkalanda -

Are you getting a specific error message?  Some more details would go a long way in helping to diagnose the issue.

Thanks!

jkalanda, 2015-04-06:
hi Jwhong 
I appreciate you getting back to me. Upon unzipping the project and then importing Mooshimester-AndroidApp-master into android studio I get the following error:

Grad;e 'Mooshimeter-AdnroidApp-master' project refresh failed
	Error:Confugration with name "default" not found

Any pointer would be appreciated. Thank you in advance

jkalanda, 2015-04-08:
Never mind!! 
Found the answer
GraphView needed to be download from  https://github.com/mooshim/GraphView and extracted in here  
Mooshimeter-AndroidApp/GraphView

Source:http://moosh.im/f/topic/build-process-android-app/

jwhong, 2015-04-08:
Hooray!  You fixed it before I could get to it.  Thanks!

closed 2016-05-06 by jwhong