mooshim/Mooshimeter-AndroidApp #17: Fails to build from source (Error: No resource found that matches the given name 'divider')

sigurg, 2015-11-02:
I just tried updating the repo and get the following error while building the app (using `gradlew assemble`):

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugResources'.
> com.android.ide.common.internal.LoggedErrorException: Failed to run command:
	/path/to/android-sdk-linux/build-tools/21.1.1/aapt package -f --no-crunch -I /path/to/android-sdk-linux/platforms/android-19/android.jar -M /path/to/Mooshimeter-AndroidApp/app/build/intermediates/manifests/full/debug/AndroidManifest.xml -S /path/to/Mooshimeter-AndroidApp/app/build/intermediates/res/debug -A /path/to/Mooshimeter-AndroidApp/app/build/intermediates/assets/debug -m -J /path/to/Mooshimeter-AndroidApp/app/build/generated/source/r/debug -F /path/to/Mooshimeter-AndroidApp/app/build/intermediates/res/resources-debug.ap_ --debug-mode --custom-package com.mooshim.mooshimeter -0 apk --output-text-symbols /path/to/Mooshimeter-AndroidApp/app/build/intermediates/symbols/debug
  Error Code:
	1
  Output:
	/path/to/Mooshimeter-AndroidApp/app/build/intermediates/res/debug/layout/fragment_scan.xml:26: error: Error: No resource found that matches the given name (at 'divider' with value '@drawable/divider').
```

According to `git bisect` 0aea1b543df63299285d49b379231835050ba815 is the first commit, that broke the build, previous versions build just fine.
Am I doing something wrong here, or are there perhaps some files missing from the github repo?

jwhong, 2015-11-02:
You are right, a file was missing.  I just pushed it in this commit
<http://t.sidekickopen27.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJN7t5XYgd0vTPW3MhNxY4Xr5cMVd_V9H56dxMwf512Tx002?t=https%3A%2F%2Fgithub.com%2Fmooshim%2FMooshimeter-AndroidApp%2Fcommit%2F0470e9196a4a493138859451751a0bf36f22576a&si=6084361541124096&pi=4b3d5f7e-974e-43ac-f6ca-f2dd8afc17bd>
.

Thank you!

sigurg, 2015-11-03:
Thanks, everything builds fine now.

closed 2015-11-03 by sigurg