mooshim/Mooshimeter-AndroidApp #19: Scan page shows negative resistances

johnp789, 2015-11-19:
In the scan page, if resistance is shown and nothing is connected to the leads, the resistance momentarily shows OVERLOAD, but then it shows values around -5600 ohms (negative values).

jwhong, 2015-11-19:
Hi John,

Can you verify the version of the app you're running?

johnp789, 2015-11-19:
I have version 1.0.17 (444) on a 2013 Moto X  with Android 4.4.4.

jwhong, 2015-11-19:
Very helpful, thanks.  One more thing:  look at the back Mooshimeter and
find the text that says "Mooshimeter Rev*" followed by a date - can you
write whether it says RevI or RevH?

johnp789, 2015-11-19:
Mooshimeter RevI 04/2015.  It's from the second manufacturing run.

jwhong, 2016-05-03:
Calculation of readings is unified, no more discrepency in beta

closed 2016-05-03 by jwhong