mooshim/Mooshimeter-AndroidApp #32: Arbitrary math

jwhong, 2016-08-21:
It would be helpful to users to have an interface for perform simple operations on meter readings in real time.  This is already implemented to some extent in math calculations like "Apparent Power" and "Thermocouple K", which simply apply formulas in the app to readings coming in from the meter, but there is no way for the user to plug in his own formulas.

607nm, 2016-09-03:
This would be really helpful.

Sembazuru, 2016-10-03:
This would be really helpful. Especially if calculating current over an external shunt. Plug in the resistance of the shunt (or copy from a resistance reading if using a large enough resistor) and have the display, graph, and logging report the calculated value. (For logging, I would probably prefer to have both the original raw reading and the calculated value recorded.)

kc0tfb, 2017-08-05:
Sorry I haven't the skill to fork the code and implement this myself, but please count me as another user that would be happy to see it done.

famo, 2017-08-06:
I'm also dearly missing the shunt support, I already wrote on the forum:
On the website the feature to measure high currents with shunts is explicitly advertised, yet there is no support at all in the app for shunts. :-(

What do I mean by support:
– For the top readout there should be an option to select “shunt”, with a popup where you then can enter the shunt type (ampere, voltage drop).
– the top readout would then display current (instead of ohm)
– one would be able to use the third readout (real power, etc.)