mooshim/Mooshimeter-AndroidApp #55: Selectable resolution

sbprojects, 2017-02-04:
I think the Mooshimeter more often than not shows too much resolution. It would be nice if the resolution was user selectable. For instance by swiping the displayed values left or right.

famo, 2017-05-25:
Yes, I'm also dearly missing a precision setting.

However implementing it as swipes would make the setting hard to find.

I think it would be better to have a third box, next to Frequency and sample settings, where one could set the precision which would then apply to all three readout (first, second and the combined one).

Background:
The sensors on the mooshi are apparently very sensitive which leads to a load of noise / constant changing values on the x.th decimal place, although nothing is really happening.
For me this is confusing when looking at the display, but it is really annoying with the voice output. Because there is constant babbling, with way too much decimal places.

When I’m measuring for 230V I don’t need any decimal places at all.
And at the car / computer, I’m fine with with 2 decimal places, I don’t need 5.

bdlow, 2017-09-27:
As a first step, it'd be great if the displayed resolution reflected the meter's accuracy in each mode/range.