mooshim/Mooshimeter-AndroidApp #64: Connection is delayed 6 seconds by ADMIN:DIAGNOSTIC:BADREAD

eldstal, 2017-08-12:
Connecting to my Mooshimeter rev1 takes about 12 seconds, and the profiler indicates that most of this is during configuration. I was curious if it could be sped up somehow (it makes the app feel unresponsive and discourages "impulsive" use of the meter).

Three of the many calls to ConfigNode.reqValue() take the full two seconds (the awaitMilli() call is never aborted by a finished read, but rather time out), which accounts for half of this time.

This happens for the three codes 16, 17 and 18 (see attached LogCat), which as far as I can tell are for LOG.INFO.[INDEX, END_TIME, N_BYTES].

I don't have an SD card to test with, in case this issue only appears for users with no log data.

LogCat output:
```
08-13 00:45:21.866 16399-16958/com.mooshim.mooshimeter D/ConfigTree: CMD: ADMIN:CRC32
08-13 00:45:22.000 16399-16983/com.mooshim.mooshimeter D/ConfigTree: ADMIN:CRC32:343497802
08-13 00:45:22.002 16399-16983/com.mooshim.mooshimeter D/ConfigTree: RECV: 6 6 bytes
08-13 00:45:22.010 16399-16958/com.mooshim.mooshimeter D/ConfigTree: ADMIN:TREE:[B@7dd0aff
08-13 00:45:22.012 16399-16958/com.mooshim.mooshimeter D/ConfigTree: java.nio.ByteArrayBuffer[position=0,limit=790,capacity=790]
08-13 00:45:22.043 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   ADMIN
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     0:CRC32
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     1:TREE
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     2:DIAGNOSTIC
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   3:PCB_VERSION
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   4:NAME
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   5:TIME_UTC
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   6:TIME_UTC_MS
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   7:BAT_V
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   8:REBOOT
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     NORMAL
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     SHIPMODE
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   SAMPLING
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     9:RATE
08-13 00:45:22.044 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       125
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       250
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       500
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       1000
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       2000
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       4000
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       8000
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     10:DEPTH
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       32
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       64
08-13 00:45:22.045 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       128
08-13 00:45:22.046 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       256
08-13 00:45:22.046 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     11:TRIGGER
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       OFF
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       SINGLE
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       CONTINUOUS
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   LOG
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     12:ON
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     13:INTERVAL
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     14:STATUS
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     15:POLLDIR
08-13 00:45:22.047 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     INFO
08-13 00:45:22.048 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       16:INDEX
08-13 00:45:22.048 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       17:END_TIME
08-13 00:45:22.048 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       18:N_BYTES
08-13 00:45:22.048 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     STREAM
08-13 00:45:22.048 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       19:INDEX
08-13 00:45:22.048 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       20:OFFSET
08-13 00:45:22.048 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       21:DATA
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:   CH1
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     22:MAPPING
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       CURRENT
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:         10
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       TEMP
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:         350
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       SHARED
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     23:RANGE_I
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:     24:ANALYSIS
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       MEAN
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       RMS
08-13 00:45:22.049 16399-16958/com.mooshim.mooshimeter D/ConfigTree:       BUFFER
[... 183 lines cut ...]
08-13 00:45:36.664 16399-16983/com.mooshim.mooshimeter D/ConfigTree: Underflow caught
08-13 00:45:36.664 16399-16983/com.mooshim.mooshimeter D/ConfigTree: RECV: 53 20 bytes
08-13 00:45:36.664 16399-16983/com.mooshim.mooshimeter D/ConfigTree: REAL_PWR:-205.66626
08-13 00:45:36.666 16399-16983/com.mooshim.mooshimeter D/ConfigTree: RECV: 54 5 bytes
08-13 00:45:37.273 16399-16983/com.mooshim.mooshimeter D/ConfigTree: TIME_UTC:1502577936
08-13 00:45:37.273 16399-16983/com.mooshim.mooshimeter D/ConfigTree: TIME_UTC_MS:55
08-13 00:45:37.273 16399-16983/com.mooshim.mooshimeter D/ConfigTree: CH1:VALUE:-1.4887503E10
08-13 00:45:37.283 16399-16983/com.mooshim.mooshimeter D/ConfigTree: CH2:VALUE:1.6487003E-4
08-13 00:45:37.295 16399-16983/com.mooshim.mooshimeter D/ConfigTree: Underflow caught
08-13 00:45:37.295 16399-16983/com.mooshim.mooshimeter D/ConfigTree: RECV: 55 20 bytes
08-13 00:45:37.313 16399-16983/com.mooshim.mooshimeter D/ConfigTree: REAL_PWR:-203.946
08-13 00:45:37.315 16399-16983/com.mooshim.mooshimeter D/ConfigTree: RECV: 56 5 bytes
08-13 00:45:37.955 16399-16983/com.mooshim.mooshimeter D/ConfigTree: TIME_UTC:1502577936
08-13 00:45:37.956 16399-16983/com.mooshim.mooshimeter D/ConfigTree: TIME_UTC_MS:697
08-13 00:45:37.956 16399-16983/com.mooshim.mooshimeter D/ConfigTree: CH1:VALUE:-1.5051107E10
08-13 00:45:37.962 16399-16983/com.mooshim.mooshimeter D/ConfigTree: CH2:VALUE:1.8318891E-4
08-13 00:45:37.967 16399-16983/com.mooshim.mooshimeter D/ConfigTree: Underflow caught
08-13 00:45:37.967 16399-16983/com.mooshim.mooshimeter D/ConfigTree: RECV: 57 20 bytes
08-13 00:45:37.968 16399-16983/com.mooshim.mooshimeter D/ConfigTree: REAL_PWR:-203.52199
08-13 00:45:37.968 16399-16983/com.mooshim.mooshimeter D/ConfigTree: RECV: 58 5 bytes
08-13 00:45:38.589 16399-16983/com.mooshim.mooshimeter D/ConfigTree: TIME_UTC:1502577937
08-13 00:45:38.589 16399-16983/com.mooshim.mooshimeter D/ConfigTree: TIME_UTC_MS:340
08-13 00:45:38.589 16399-16983/com.mooshim.mooshimeter D/ConfigTree: CH1:VALUE:-1.5341907E10
08-13 00:45:38.592 16399-16983/com.mooshim.mooshimeter D/ConfigTree: CH2:VALUE:1.8318891E-4
```

famo, 2017-08-13:
I have an SD-Card in mine. If you tell me how you created the log I can try here.

eldstal, 2017-08-13:
Connect using ADB, build the app and select "run" in android studio, open the "Android Monitor" tab at the bottom of the window and add the search filter "ConfigTree" (there's a _lot_ of debugging output).

This capture was made by opening the menu, connecting to the meter and then disconnecting as soon as it was finished.

famo, 2017-08-17:
How do you build the app?
(I'm under linux btw)

eldstal, 2017-08-17:
Sorry, should have gone into more detail.

You will need Android studio (probably available in your package manager) as well as the Android SDK and build tools, etc (should all come in as dependencies of android-studio, no worries).

Check the app out from github and load it as a project in android-studio.

First time around, it will fail to rebuild (it tries automatically as far as I can remember), and you will get messages saying stuff like "Gradle update failed" and a little link with a proposed solution (stuff like "Update build tools to 2.3.2" or "install newer Android SDK"). Keep clicking the links to get your environment set up and eventually the build will complete without errors.

When it has built successfully, enable ADB debugging on your android device and connect it to the computer via USB (it can be done over wifi too, if you google around a little). Hit the "run" button in the tool bar of android studio to push the newly built application to the phone and run it. You will get a warning about signing and invalid app version, saying the currently installed Mooshimeter app needs to be uninstalled. Let it do that.

Once it's up and running, you can see the log in the bottom of the android studio window, under "Android Monitor" (I think it's tab number 6) and it has a "Logcat" tab.

You can filter the log by the string "ConfigTree" to cut down on the noise.

eldstal, 2017-08-18:
OK, I've tried it with an ill-formatted SD-card (same problem), a properly formatted but empty card (same problem) and finally, a card with some log data on it.

here is what happens with log data present on the card:

```
08-18 21:53:53.834 21656-21940/com.mooshim.mooshimeter D/ConfigTree: CMD: ADMIN:CRC32
08-18 21:53:53.932 21656-21960/com.mooshim.mooshimeter D/ConfigTree: ADMIN:CRC32:343497802
08-18 21:53:53.932 21656-21960/com.mooshim.mooshimeter D/ConfigTree: RECV: 211 6 bytes
08-18 21:53:53.932 21656-21940/com.mooshim.mooshimeter D/ConfigTree: ADMIN:TREE:[B@330d250
08-18 21:53:53.933 21656-21940/com.mooshim.mooshimeter D/ConfigTree: java.nio.ByteArrayBuffer[position=0,limit=790,capacity=790]
08-18 21:53:53.940 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   ADMIN
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     0:CRC32
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     1:TREE
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     2:DIAGNOSTIC
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   3:PCB_VERSION
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   4:NAME
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   5:TIME_UTC
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   6:TIME_UTC_MS
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   7:BAT_V
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   8:REBOOT
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     NORMAL
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     SHIPMODE
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   SAMPLING
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     9:RATE
08-18 21:53:53.941 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       125
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       250
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       500
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       1000
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       2000
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       4000
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       8000
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     10:DEPTH
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       32
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       64
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       128
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       256
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     11:TRIGGER
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       OFF
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       SINGLE
08-18 21:53:53.942 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       CONTINUOUS
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   LOG
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     12:ON
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     13:INTERVAL
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     14:STATUS
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     15:POLLDIR
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     INFO
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       16:INDEX
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       17:END_TIME
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       18:N_BYTES
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     STREAM
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       19:INDEX
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       20:OFFSET
08-18 21:53:53.943 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       21:DATA
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:   CH1
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     22:MAPPING
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       CURRENT
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:         10
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       TEMP
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:         350
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       SHARED
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     23:RANGE_I
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     24:ANALYSIS
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       MEAN
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       RMS
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:       BUFFER
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     25:VALUE
08-18 21:53:53.944 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     26:OFFSET
08-18 21:53:53.945 21656-21940/com.mooshim.mooshimeter D/ConfigTree:     27:BUF
[... cut ...]
08-18 21:54:04.227 21656-21960/com.mooshim.mooshimeter D/ConfigTree: RECV: 6 20 bytes
08-18 21:54:04.227 21656-21960/com.mooshim.mooshimeter D/ConfigTree: REAL_PWR:533.28564
08-18 21:54:04.228 21656-21960/com.mooshim.mooshimeter D/ConfigTree: RECV: 7 5 bytes
08-18 21:54:04.495 21656-21656/com.mooshim.mooshimeter D/ConfigTree: CMD: PCB_VERSION
08-18 21:54:04.608 21656-21960/com.mooshim.mooshimeter D/ConfigTree: PCB_VERSION:8
```

It appears that the timing behavior is the same, although more communication takes place during the BADREADs.