mooshim/Mooshimeter-AndroidApp #66: App still maintained?

IzzySoft, 2019-02-22:
Is the app still maintained as open-source? I wonder as the last tag is release_37 from 11/2017 and no commit since 8/2017 (and the last APK in the repo is release_32 from 9/2016) – while Playstore has a version 38 from 1/2018. So did your app go commercial, and is no longer open source? Will we see at least the v37 APK here?

AndrewSmart, 2019-03-09:
Did you see the [beta](https://github.com/mooshim/Mooshimeter-AndroidApp/tree/beta) branch?

IzzySoft, 2019-03-09:
Not until now :see_no_evil: But still: last commit there was in 5/2018 (also almost a year ago already).

IzzySoft, 2019-12-16:
@jwhong any statement on this? I see you've meanwhile added a privacy policy, which a.o. states that you collect logs to improve the app. That at least sounds as if you consider picking up development again?